cedricxie/apollo_perception_ros #1: Unknown runtime specified nvidia.

yxf010, 2019-04-09:
when run **bash -x docker/run.sh**
+ '[' '' = '' ']'
+ DOCKER_NAME=cedricxie/apollo-perception-ros:latest
+ XSOCK=/tmp/.X11-unix
+ XAUTH=/home/phe/.Xauthority
+ SHARED_DIR=/home/yx/shared_dir
+ HOST_DIR=/home/phe/shared_dir
+ mkdir -p /home/phe/shared_dir
+ echo 'Shared directory: /home/phe/shared_dir'
Shared directory: /home/phe/shared_dir
+ docker run -it --rm --volume=/home/phe/shared_dir:/home/yx/shared_dir:rw --volume=/tmp/.X11-unix:/tmp/.X11-unix:rw --volume=/home/phe/.Xauthority:/home/phe/.Xauthority:rw --env=XAUTHORITY=/home/phe/.Xauthority --env=DISPLAY=:0 --env=QT_X11_NO_MITSHM=1 -u yx --privileged -v /dev/bus/usb:/dev/bus/usb -v /dev/video0:/dev/video0 --net=host --runtime=nvidia cedricxie/apollo-perception-ros:latest
docker: **Error response from daemon: Unknown runtime specified nvidia.**
See 'docker run --help'.

  error :--runtime=nvidia \ 
how to slove it?

cedricxie, 2019-04-09:
Hello, please install nvidia docker.

yxf010, 2019-04-09:
[build] Summary: 7 of 8 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  1 packages succeeded with warnings.                       
[build]   Abandoned: None.                                                     
[build]   **Failed:    1 packages failed.**                                        
[build] Runtime: 17 minutes and 38.9 seconds total.                            
[build] Note: Workspace packages have changed, please re-source setup files to use them.
are you chinese?My wechat is yxf010

cedricxie, 2019-04-10:
Hello @yxf010 , the rosbag is the recorded sensor data. I would suggest to go through the ros tutorial at http://wiki.ros.org/ROS/Tutorials to get yourself familiar with the framework. Thank you.

Yuesong

yasenh, 2019-04-11:
@yxf010 
I think the building error was mentioned in "[Known issues](https://github.com/cedricxie/apollo_perception_ros#known-issues)", just rerun the `catkin build`

jinjinhong, 2019-04-15:
@cedricxie I have installed nvidia docker,it does not work .docker: Error response from daemon: Unknown runtime specified nvidia. I don't know why.Have you met?