cedricxie/apollo_perception_ros #2: protobuf error

muzi2045, 2019-04-11:
Trying compile the whole code in the single ros-kinetic environment .
Got some the libprotobuf error:
```
[100%] Building CXX object perception/apollo_perception_ros/src/CMakeFiles/perception_yx_node.dir/perception_yx.cpp.o
[100%] Linking CXX executable /home/aisimba/perception_ws/devel/lib/apollo_perception_ros/perception_yx_node
/usr/bin/ld: warning: libboost_system.so.1.54.0, needed by /home/aisimba/Documents/caffe_x86/output-GPU/lib/libcaffe.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_thread.so.1.54.0, needed by /home/aisimba/Documents/caffe_x86/output-GPU/lib/libcaffe.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libboost_filesystem.so.1.54.0, needed by /home/aisimba/Documents/caffe_x86/output-GPU/lib/libcaffe.so, not found (try using -rpath or -rpath-link)
/usr/bin/ld: warning: libprotobuf.so.13, needed by /home/aisimba/Documents/caffe_x86/output-GPU/lib/libcaffe.so, may conflict with libprotobuf.so.9
/usr/bin/ld: /home/aisimba/perception_ws/devel/lib/libroi_filter_proto_lib.so: undefined reference to symbol '_ZN6google8protobuf8internal16RegisterAllTypesEPKNS0_8MetadataEi'
/usr/local/lib/libprotobuf.so.13: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
perception/apollo_perception_ros/src/CMakeFiles/perception_yx_node.dir/build.make:536: recipe for target '/home/aisimba/perception_ws/devel/lib/apollo_perception_ros/perception_yx_node' failed
make[2]: *** [/home/aisimba/perception_ws/devel/lib/apollo_perception_ros/perception_yx_node] Error 1
CMakeFiles/Makefile2:5177: recipe for target 'perception/apollo_perception_ros/src/CMakeFiles/perception_yx_node.dir/all' failed
make[1]: *** [perception/apollo_perception_ros/src/CMakeFiles/perception_yx_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```
seems like the libroi_filter_proto_lib.so have some linking problem, hope for any suggestions!
@cedricxie

cedricxie, 2019-04-11:
Hello @muzi2045 , are you doing the catkin build inside docker container? Looks like some libraries are not found.

muzi2045, 2019-04-11:
not in the docker. just compile it in a new ROS environment.

yasenh, 2019-04-11:
@muzi2045 
Seems you have a older version of protobuf installed(libprotobuf.so.9), but a newer version is required (libprotobuf.so.13). 
I think you can try to modify the CmakeLists locally:
https://github.com/cedricxie/apollo_perception_ros/blob/master/src/perception/apollo_perception_ros/CMakeLists.txt#L64-L66

find_package(Protobuf 3.3 REQUIRED EXACT) -> find_package(Protobuf **2.6** REQUIRED EXACT) , which is the corresponding version of libprotobuf.so.9.

However, I still recommend you to use docker instead :)

muzi2045, 2019-04-12:
I have installed the protobuf 3.3.0 in the system, And cmake can find it in the /usr/local/lib
which confuse me is the libcaffe.so and libproto.a file, I can't get which version protobuf it built with.
I'll try to run it in the Docker.
```
/usr/local/bin/cmake -DCMAKE_BUILD_TYPE=Debug -G "CodeBlocks - Unix Makefiles" /home/aisimba/aisimba_ws/src/perception_ros
-- Using CATKIN_DEVEL_PREFIX: /home/aisimba/aisimba_ws/src/perception_ros/cmake-build-debug/devel
-- Using CMAKE_PREFIX_PATH: /home/aisimba/aisimba_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/aisimba/aisimba_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/aisimba/aisimba_ws/src/perception_ros/cmake-build-debug/test_results
-- Found gmock sources under '/usr/src/gmock': gmock will be built
-- Found gtest sources under '/usr/src/gmock': gtests will be built
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.14
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Boost version: 1.58.0
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.2/VTKTargets.cmake"
but not all the files it references.

-- Could NOT find PCL_APPS (missing: PCL_APPS_LIBRARY) 
-- Found Protobuf: /usr/local/lib/libprotobuf.so;-lpthread (found suitable exact version "3.3.0") 
Protobuf Include path: /usr/local/include
Protobuf libraries: /usr/local/lib/libprotobuf.so-lpthread
Protobuf library: /usr/local/lib/libprotobuf.so
-- Configuring done
-- Generating done
```

cedricxie, 2019-04-16:
Thanks for answering! @yasenh

closed 2021-03-03 by muzi2045