cedricxie/apollo_perception_ros #3: can not build

jinjinhong, 2019-04-15:
@cedricxie hi,I have done step by step ,but enter the docker ,the inside does not have anything,can not use the command catkin build,I don't know the reson

jinjinhong, 2019-04-15:
I was very interested, and I didn't touch Apollo very long.can you give me more information about how to run on my own computer? thank you ,my email is jinjhong1027@163.com

cedricxie, 2019-04-15:
Hello @jinjinhong , did you see shared_dir folder in your home directory in the docker?

jinjinhong, 2019-04-15:
yes,but run the catkin build command ,error like this
CMake Error at /home/yx/shared_dir/apollo_perception_ros/build/catkin_tools_prebuild/CMakeLists.txt:12 (message):
  The catkin CMake module was not found, but it is required to build a linked
  workspace.  To resolve this, please do one of the following, and try
  building again.

   1. Source the setup.sh file from an existing catkin workspace:
      source SETUP_FILE

   2. Extend another catkin workspace's result (install or devel) space:
      catkin config --extend RESULT_SPACE

   3. Set `catkin_DIR` to the directory containing `catkin-config.cmake`:
      catkin config --cmake-args -Dcatkin_DIR=CATKIN_CMAKE_CONFIG_PATH

   4. Add the catkin source package to your workspace's source space:
      cd SOURCE_SPACE && git clone https://github.com/ros/catkin.git

cd /home/yx/shared_dir/apollo_perception_ros/build/catkin_tools_prebuild; catkin build --get-env catkin_tools_prebuild | catkin env -si  /usr/bin/cmake /home/yx/shared_dir/apollo_perception_ros/build/catkin_tools_prebuild --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/yx/shared_dir/apollo_perception_ros/devel/.private/catkin_tools_prebuild -DCMAKE_INSTALL_PREFIX=/home/yx/shared_dir/apollo_perception_ros/install; cd -
.......................................................................................................................................................................................
Failed     << catkin_tools_prebuild:cmake          [ Exited with code 1 ]                                                                                                              
Failed    <<< catkin_tools_prebuild                [ 0.0 seconds ]                                                                                                                     
Abandoned <<< geometry_msgs                        [ Unrelated job failed ]                                                                                                            
Abandoned <<< pcl_msgs                             [ Unrelated job failed ]                                                                                                            
Abandoned <<< pcl_conversions                      [ Unrelated job failed ]                                                                                                            
Abandoned <<< pcl_ros                              [ Unrelated job failed ]                                                                                                            
Abandoned <<< apollo_perception_ros                [ Unrelated job failed ]                                                                                                            
Abandoned <<< vehicle_base                         [ Unrelated job failed ]                                                                                                            
Abandoned <<< visualization_msgs                   [ Unrelated job failed ]                                                                                                            
[build] Summary: 0 of 8 packages succeeded.                                                                                                                                            
[build]   Ignored:   None.                                                                                                                                                             
[build]   Warnings:  None.                                                                                                                                                             
[build]   Abandoned: 7 packages were abandoned.                                                                                                                                        
[build]   Failed:    1 packages failed.                                                                                                                                                
[build] Runtime: 0.1 seconds total.                                                                                                                                                    
yx@hjj-ZHAOYANG-K42-80:~/shared_dir/apollo_perception_ros$

cedricxie, 2019-04-15:
Hi @jinjinhong , can you try "source /opt/ros/indigo/setup.bash" inside docker to see if it helps?

jinjinhong, 2019-04-15:
I have try many times ,it does not work.

cedricxie, 2019-04-16:
Hello @jinjinhong , it is complaining the CMAKE module is not found. But I think it should already be installed in the ROS environment in the docker.

If you want, you can try installing the CMAKE module to see if it helps. But I am not sure what's going wrong here.

jinjinhong, 2019-04-16:
Thanks,I still have a question. When catkin build,Is there any good way to solve such a mistake?
 Could not find a package configuration file provided by "roscpp" with any
  of the following names:
    roscppConfig.cmake
    roscpp-config.cmake
 Could not find a package configuration file provided by
  "message_generation" with any of the following names:
    message_generationConfig.cmake
    message_generation-config.cmake

jinjinhong, 2019-04-16:
Will it be a problem with docker image?

cedricxie, 2019-04-16:
@jinjinhong , I tested the docker image on my side and it was working. You can check out the build.sh script and build the image yourself. I would suggest to try everything from the beginning again and see if you notice anything abnormal.

yasenh, 2019-04-16:
@jinjinhong Current docker images works for me as well, and I am using Ubuntu 16.04

jinjinhong, 2019-04-16:
I using 14.04 and  do many times,but meet this question: Could not find a package configuration file provided by "roscpp"

closed 2026-06-29 by cedricxie