cedricxie/apollo_perception_ros #6: A probability issue relate with CUDA10

jinfagang, 2019-05-08:
I know cuda10 is new but since I am using tensorflow 2.0 apis so move on cuda10 is the eventually things, this codes now can not built on cuda10, it has this error:

```
apollo_perception_ros/src/cuda_util/util.cu(155): error: incomplete type is not allowed
```

this file, this line, and only 1 error. Does this can be fix for cuda10 compatible?

cedricxie, 2019-06-09:
Hello @jinfagang , I haven't tried cuda 10, but I have doubts about it since the static models that come with apollo might not be easily adjusted to a newer version of cuda.

CodingSmith, 2020-06-26:
> I know cuda10 is new but since I am using tensorflow 2.0 apis so move on cuda10 is the eventually things, this codes now can not built on cuda10, it has this error:
>
> ``` [...]

I met the same issue with CUDA10, and I solve this issue by adding below into util.cu,

#include <opencv2/core/core.hpp>
#include <opencv2/opencv.hpp>

Hopefully, you problem could be solved as well.

liuqf0425, 2021-07-08:
@CodingSmith  @jinfagang I met the same error too, and the error was indeed related to OpenCV. After including the heading files of opencv, the error would no more occurs.