cedricxie/apollo_perception_ros #8: an error when runing sudo ./run.sh

firesnow1234, 2019-08-11:
Linux 16.0.4
Docker 19.03.1
NVIDIA Docker: 1.0.1

(base) ivlab@ivlab:~/shared_dir/apollo_perception_ros$ sudo ./docker/run.sh
Shared directory: /home/root/shared_dir
docker: **Error response from daemon: OCI runtime create failed: unable to retrieve OCI runtime error** (open /run/containerd/io.containerd.runtime.v1.linux/moby/1598507c8c9d8bb6a2f94748b4edec40ffc51bd2458dfc473f7d395931596188/log.json: no such file or directory): fork/exec /usr/bin/nvidia-container-runtime: no such file or directory: : unknown.

Do you have any idea about this problem?

cedricxie, 2019-08-11:
Hi @firesnow1234 , I only tested with nvidia-docker 2. You probably have to make some changes with nvidia-docker 1. I haven't seen this error myself.

firesnow1234, 2019-08-12:
Thanks for your reply, does there have any requirements for the version of docker? Can you share the requirements of the running environment？

closed 2019-08-13 by firesnow1234

NoRomantic, 2019-12-03:
Hello, have you solved this problem yet?

firesnow1234, 2019-12-09:
> Hello, have you solved this problem yet?

Yes, I installed nvidia-docker2, you probably need to install it.