cedricxie/apollo_perception_ros #14: Running with ROS Kinetic

argh1, 2019-10-28:
Is there a way to get this running under ROS Kinetic? I modified the docker scripts to use Ubuntu 16.04 and install ROS Kinetic, but I'm having build issues, seemingly around PCL. It looks like Kinetic or its dependencies is using vtk6 but this build is reliant upon vtk5. Does anyone know what modifications I could make to get this to work with ROS Kinetic?

Thank you!

NoRomantic, 2019-12-02:
Hello, have you solved this problem please?

HTZfighter1, 2021-06-24:
您好，请问源代码中是否也使用了大陆毫米波雷达作为融合呢， 因为看到大面积的代码被注释掉了。 有一些疑惑， 时候使用雷达融合时候存在问题，多谢。
void PerceptionYX::processRadarData(const apollo_perception_standalone::ContiRadar &msg) {

HTZfighter1, 2021-06-24:
hello I use the 1604+cuda8+gefore1050TI for the build. 
相同配置可以交流一下