cedricxie/apollo_perception_ros #18: permute_layer.cu:47] Check failed: error == cudaSuccess (8 vs. 0)  invalid device function

liuqf0425, 2021-06-03:
Thank you for this project, it really helps. When  running the node, an error would always occurred.  The GPU device on My host machine is  NVIDIA RTX2060 , and driver version is  460.73.01, UBUNTU 18.04. 

please check:

auto-starting new master
process[master]: started with pid [16304]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a6e39dce-c457-11eb-84b4-3c9c0fb840d4
process[rosout-1]: started with pid [16317]
started core service [/rosout]
process[perception_yx_node-2]: started with pid [16334]
process[robot_state_publisher-3]: started with pid [16343]
process[joint_state_publisher-4]: started with pid [16344]
WARNING: Logging before InitGoogleLogging() is written to STDERR
W0603 10:37:11.385072 16334 perception_yx.cpp:51] Sensor Lidar: Enabled, Logging: False
W0603 10:37:11.385154 16334 perception_yx.cpp:55] Sensor Radar: Disabled, Logging: False
W0603 10:37:11.385161 16334 perception_yx.cpp:59] Sensor Camera Long: Enabled, Logging: True
W0603 10:37:11.385167 16334 perception_yx.cpp:63] Sensor Camera Short: Enabled, Logging: True
W0603 10:37:11.385174 16334 perception_yx.cpp:67] Fusion, Logging: True
W0603 10:37:11.385179 16334 perception_yx.cpp:69] Image Resize Factor: 1
W0603 10:37:11.390164 16334 perception_yx.cpp:83] Parse camera calibration : success
process[rviz-5]: started with pid [16406]
.....
.....
Warning: TF_OLD_DATA ignoring data from the past for frame novatel at time 1.51381e+09 according to authority unknown_publisher  Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained  at line 277 in /tmp/binarydeb/ros-indigo-tf2-0.5.20/src/buffer_core.cpp
F0603 10:33:18.607558 15890 permute_layer.cu:47]  Check failed: error == cudaSuccess (8 vs. 0)  invalid device function
*** Check failure stack trace: ***

HTZfighter1, 2021-06-24:
您好，请问源代码中是否也使用了大陆毫米波雷达作为融合呢， 因为看到大面积的代码被注释掉了。 有一些疑惑， 时候使用雷达融合时候存在问题，多谢。
void PerceptionYX::processRadarData(const apollo_perception_standalone::ContiRadar &msg) {

Findurname, 2021-09-07:
For this error: `Check failed: error == cudaSuccess (8 vs. 0) invalid device function`,  is caused by the RTX2060. 

The docker's caffe lib does not support RTX2060. But 1070, 1080 and 1080ti will be OK.